chipsalliance/fpga-interchange-tests #3: Add Kokoro CI

acomodi, 2021-04-16:
With the fasm2bels flow enabled by https://github.com/SymbiFlow/fpga-interchange-tests/pull/2, we require a CI with vendor tools to run the tests to the end of the flow, namely vendor tool bitstream generation.

acomodi, 2021-04-16:
cc @mithro

mithro, 2021-04-29:
```
========================================
Git version info
----------------------------------------
commit 787c404a78b65aeaee2caaf44130bb3edf188a23
Merge: ea33aaf 138feef
Author: Tim Ansell <tansell@google.com>
Date:   Thu Apr 29 07:13:23 2021 -0700

    Merge pull request #4 from antmicro/add-kokoro-ci
    
    Add kokoro ci
----------------------------------------
fatal: No names found, cannot describe anything.
----------------------------------------
787c404
----------------------------------------

===========================================
Running FPGA interchange tests
-------------------------------------------
make: *** No rule to make target 'all-xc7-tests'.  Stop.

[ID: 4676367] Build finished after 292 secs, exit value: 1

Warning: Permanently added 'localhost' (ECDSA) to the list of known hosts.
```

acomodi, 2021-04-29:
Should get fixed with https://github.com/SymbiFlow/fpga-interchange-tests/pull/12. Not sure why the presubmit CI is not starting though.

mithro, 2021-04-29:
I forgot to add the webhook, seems to be working.

mithro, 2021-04-29:
Seems like env is getting saved?
```
[09:15:48] Storing build artifacts to Placer
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/downloads/conda-pkgs/python-3.7.10-hdb3f193_0/info/about.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/downloads/conda-pkgs/python-3.7.10-hdb3f193_0/info/about.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/conda/pkgs/cache/2ce54b42.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/conda/pkgs/cache/2ce54b42.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/downloads/conda-pkgs/prjxray-db-0.0_248_g2e51ad3-20210312_125539/share/symbiflow/prjxray-db/zynq7/tile_type_CFG_SECURITY_BOT_PELE1.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/downloads/conda-pkgs/prjxray-db-0.0_248_g2e51ad3-20210312_125539/share/symbiflow/prjxray-db/zynq7/tile_type_CFG_SECURITY_BOT_PELE1.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/conda/pkgs/ncurses-6.2-he6710b0_1/info/index.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/conda/pkgs/ncurses-6.2-he6710b0_1/info/index.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/conda/envs/fpga-interchange/conda-meta/bzip2-1.0.8-h7b6447c_0.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/conda/envs/fpga-interchange/conda-meta/bzip2-1.0.8-h7b6447c_0.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/downloads/conda-pkgs/prjxray-db-0.0_248_g2e51ad3-20210312_125539/share/symbiflow/prjxray-db/zynq7/tile_type_CFG_CENTER_BOT.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/downloads/conda-pkgs/prjxray-db-0.0_248_g2e51ad3-20210312_125539/share/symbiflow/prjxray-db/zynq7/tile_type_CFG_CENTER_BOT.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/conda/envs/fpga-interchange/share/symbiflow/prjxray-db/artix7/tile_type_GTP_INT_INT_TERM_R.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/conda/envs/fpga-interchange/share/symbiflow/prjxray-db/artix7/tile_type_GTP_INT_INT_TERM_R.json
[09:17:46] Copied /tmp/workspace/workspace/foss-fpga-tools/fpga-interchange-tests/presubmit/artifacts/github/fpga-interchange-tests/env/downloads/conda-pkgs/pcre-8.44-he6710b0_0/info/paths.json to /placer/prod/scratch/home/kokoro-dedicated/build_artifacts/prod/foss-fpga-tools/fpga-interchange-tests/presubmit/1/20210429-090351/env/downloads/conda-pkgs/pcre-8.44-he6710b0_0/info/paths.json
```

acomodi, 2021-04-29:
> Seems like env is getting saved?

Indeed, the .json files in the env directory end up in the bucket as well. I'll fix this.

acomodi, 2021-05-24:
Kokoro is now active. Closing.

closed 2021-05-24 by acomodi