chipsalliance/fpga-interchange-tests #40: Need to specify all cell's IOs for correct DCP

acomodi, 2021-06-24:
When explicitly instantiating a cell, all the IOs need to be explicitly stated, or otherwise an invalid DCP is generated, where the cell results to be unplaced, despite being present in the physical netlist.

I will provide a small and reproducible test case soon to showcase the issue.

gatecat, 2021-06-24:
If we specify the [default cell connections](https://github.com/SymbiFlow/fpga-interchange-schema/blob/main/interchange/DeviceResources.capnp#L435-L453) then this should be avoidable, as nextpnr should create and connect these ports automatically.

RapidWright isn't currently generating these and I don't think it has data for those, but perhaps this is something that could be added (I demonstrated how Vivado handles this at https://github.com/SymbiFlow/fpga-interchange-schema/pull/35#issuecomment-819709256 )

acomodi, 2021-06-24:
Ok, thanks for the pointers.

In case RW cannot get the required data, I wonder whether it might be a valid (and feasible) solution to patch the device? It may be quite verbose to add a patch for all the cells, but I cannot think of another way of providing this data.