chipsalliance/fpga-interchange-tests #56: Issue with the rapidyaml Python package

mkurc-ant, 2021-07-30:
This issue is to track `rapidyaml` package import errors, another issue has been posted in the rapidyaml repo: https://github.com/biojppm/rapidyaml/issues/146

With the PR https://github.com/SymbiFlow/fpga-interchange-tests/pull/42 I've noticed kokoro CI failures on `rapidyaml` import. whereas the GH actions CI imports it successfully even though it uses the same Conda environmeng.

By attaching the following debug command just before invoking the tests themselves to each CI script:
```
strace python -c "import ryml" 2>&1 | grep "\.so" || true
```
I discovered that the Python interpreter always tries to link `libpython3.9.so.1.0` even though the conda env uses Python 3.7. 

On GH actions the import succeeds because the library is found in the system **outside the conda env** while on kokoro it is not.

mithro, 2021-07-30:
@mkurc-ant - Shouldn't conda package building be complaining about that?

mkurc-ant, 2021-08-02:
@mithro We install rapidyaml through pip, not conda. I haven't been fully through the package build system but it looks like the package has the requirement of Python >= 3.7