chipsalliance/fpga-interchange-tests #94: fasm2bels tests failure to read DCP in Vivado

acomodi, 2022-02-24:
With the RapidWright bump done in https://github.com/SymbiFlow/fpga-interchange-tests/pull/90, the fasm2bels tests have started to fail due to an error when reading the fasm2bels-generated DCP back into Vivado:

```
Time (s): cpu = 00:00:00.02 ; elapsed = 00:00:00.02 . Memory (MB): peak = 1060.848 ; gain = 0.000 ; free physical = 45857 ; free virtual = 67196                                                                                                                                          
INFO: [Netlist 29-17] Analyzing 20 Unisim elements for replacement
INFO: [Netlist 29-28] Unisim Transformation completed in 0 CPU seconds                                                                       
INFO: [Project 1-479] Netlist was created with RapidWright 2021.2.1
INFO: [Device 21-403] Loading part xc7a35tcpg236-1                                                                                           
INFO: [Project 1-570] Preparing netlist for logic optimization                                                                               
Parsing XDC File [/data/interchange/fpga-interchange-tests/build/tests/designs/counter/basys3/.Xil/Vivado-3299046-acomodi/dcp3/null.xdc]
WARNING: [Vivado 12-507] No nets matched '<const0>'. [/data/interchange/fpga-interchange-tests/build/tests/designs/counter/basys3/.Xil/Vivado-3299046-acomodi/dcp3/null.xdc:144]                                                                                                          
CRITICAL WARNING: [Common 17-55] 'set_property' expects at least one object. [/data/interchange/fpga-interchange-tests/build/tests/designs/counter/basys3/.Xil/Vivado-3299046-acomodi/dcp3/null.xdc:146]                                                                                  
Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.
WARNING: [Vivado 12-507] No nets matched '<const1>'. [/data/interchange/fpga-interchange-tests/build/tests/designs/counter/basys3/.Xil/Vivado-3299046-acomodi/dcp3/null.xdc:147]                                                                                                          
CRITICAL WARNING: [Common 17-55] 'set_property' expects at least one object. [/data/interchange/fpga-interchange-tests/build/tests/designs/counter/basys3/.Xil/Vivado-3299046-acomodi/dcp3/null.xdc:149]                                                                                  
Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.             
Finished Parsing XDC File [/data/interchange/fpga-interchange-tests/build/tests/designs/counter/basys3/.Xil/Vivado-3299046-acomodi/dcp3/null.xdc]                                                                                                                                         
Reading XDEF placement.                
Reading placer database...                 
ERROR: [Common 17-49] Internal Data Exception: The netlist cache file is corrupted. Expected (section_tag_instance): 0xdeadbbbb found: 0     
INFO: [Common 17-206] Exiting Vivado at Thu Feb 24 14:41:45 2022...
```

Here is a sample DCP that causes the above error: [counter.fasm2bels.dcp.gz](https://github.com/SymbiFlow/fpga-interchange-tests/files/8132900/counter.fasm2bels.dcp.gz)

To fully reproduce (from 754802d561d34cd0e1c2a0095d601b39688bb397):

```
make env
make update
make build

cd build
make xc7-counter-basys3-fasm2bels-diff-fasm
```

eddieh-xlnx, 2022-02-24:
Do you have a copy of `counter.fasm2bels.dcp` before #90?

acomodi, 2022-02-24:
@eddieh-xlnx Sure, fortunately CI has kept them, here is one produced by #80: [counter.fasm2bels_before90.dcp.zip](https://github.com/SymbiFlow/fpga-interchange-tests/files/8135105/artifacts_prod_foss-fpga-tools_fpga-interchange-tests_continuous_52_20211014-072931_build_tests_designs_counter_basys3_counter.fasm2bels.zip)

eddieh-xlnx, 2022-02-25:
I can confirm that this was a bug introduced by the latest RapidWright release and I have a fix undergoing review.

eddieh-xlnx, 2022-03-14:
This bug should now be fixed on RapidWright master. Please try it and let me know!

P.S. You should move this line:
https://github.com/chipsalliance/fpga-interchange-tests/blob/96db0e15e12c1b6cd82efcfe91045d8734839816/Makefile#L27
to:
`./gradlew updateJars`
because we pull dependencies in from Maven now instead of downloading a big jar file.

closed 2022-03-18 by acomodi