chipsalliance/fpga-interchange-tests #102: enable a public GCP bucket to store the device data and chipdb

acomodi, 2022-03-04:
A publicly accessible bucket is required to store all the device data and techmaps generated by CI, in order to have a ready-to-use database to allow faster usage of the interchange format in the various downstream toolchains. 

This is required to land https://github.com/chipsalliance/fpga-interchange-tests/pull/101, which aims at generating and packaging the required data and send it to GCP.

Once the GCP bucket is enabled and the secrets are added to this repository, each new CI run will swap the current "latest" artifacts with the newly produced, while also creating a copy to archive with information on the generation date, so to keep track of past builds outputs.

acomodi, 2022-03-04:
cc @mithro @kgugala

mithro, 2022-03-04:
I don't think we need to do the swapping stuff? Just maintain a file which redirects to the latest version?

mithro, 2022-03-04:
The bucket `fpga-interchange-tests` already exists with public read access.

mithro, 2022-03-04:
I think we want to use https://docs.github.com/en/actions/deployment/security-hardening-your-deployments/configuring-openid-connect-in-google-cloud-platform ?
> This guide gives an overview of how to configure GCP to trust GitHub's OIDC as a federated identity, and includes a workflow example for the [google-github-actions/auth](https://github.com/google-github-actions/auth) action that uses tokens to authenticate to GCP and access resources.

mithro, 2022-03-04:
I'm going to see if I can get the OIDC stuff setup.