chipsalliance/fpga-interchange-tests #125: [nextpnr_fpga_interchange] Failure in placement

clavin-xlnx, 2022-07-25:
I'm getting an `ERROR: Placing design failed.` when trying to place the attached design:

[example.zip](https://github.com/chipsalliance/fpga-interchange-tests/files/9183731/example.zip)

Log is included, but I'm also seeing several warnings prior to the ERROR message:

```
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/DYNAMIC_DCI_TS' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/IBUF_DISABLE' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/IO' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/KEEPER_EN' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/LVDS_TRUE' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OP' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OSC0' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OSC1' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OSC2' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OSC3' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OSC_EN' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OUTB_B_IN' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/PD_EN' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/PU_EN' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/TSDI' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/TSP' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/TSTATE_IN' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/VREF' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/DOUT' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/I' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/OUTB_B' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/O_B' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/SWITCH_OUT' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/TSTATEB' (no cell)
Warning: post-placement validity check failed for Bel 'IOB_X2Y272.HPIOB_SNGL/TSTATE_OUT' (no cell)
ERROR: Placing design failed.
136 warnings, 1 error
```

This is currently a blocking issue for me.

clavin-xlnx, 2022-07-25:
The command I used to run nextpnr is:

```
nextpnr-fpga_interchange --chipdb fpga-interchange-tests/build/devices/xczu7ev/xczu7ev.bin --netlist example/gnl_4_8_7_2.0_gnl_8000_08_7_80_80.netlist --phys example/gnl_4_8_7_2.0_gnl_8000_08_7_80_80.phys --package ffvc1156 --log example/gnl_4_8_7_2.0_gnl_8000_08_7_80_80.log --disable-lut-mapping-cache --xdc example/gnl_4_8_7_2.0_gnl_8000_08_7_80_80.xdc
```

mkurc-ant, 2022-07-26:
@clavin-xlnx Could you try running nextpnr additionally with `--debug` option? That may give more info on why it is failing.
Is the attached DCP the source for obtaining the logical netlist with RapidWright?

I don't see any IOB cells in the DCP when I open it in Vivado. Maybe that is the issue? If they are none then nextpnr fails.

clavin-xlnx, 2022-07-27:
Thanks @mkurc-ant, yes, the DCP included was the source.

I found you are correct in that the design was missing IO cells as it came from a design that was synthesized out of context.  When I use Vivado to add the cells and export to the Interchange, placement and routing succeeds (with some other minor issues which I will file separately).

closed 2022-07-27 by clavin-xlnx